Xinyuan-LilyGO/LilyGo-Camera-Series #2: JTAG Debugging

nest-don, 2022-08-16:
Hello, 

Is it possible to debug TTGO T-Camera Plus ESP32-DOWDQ6 using ESP-Prog?   If so how would you connect the JTAG pins to the board?  

Thanks

liangyingy, 2022-08-23:
All io has been used, jtag cannot be used.

closed 2022-08-23 by liangyingy

Jason2866, 2022-09-30:
@nest-don @liangyingy The answer is wrong. You can use the inbuilt jtag interface via the usb-c connector. No additional hardware needed. https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html#jtag-debugging-selecting-jtag-adapter

nest-don, 2022-10-05:
I think the inbuilt jtag is only available in the new ESP32-C3.

nest-don, 2022-10-06:
So the question relates to this product -> http://www.lilygo.cn/prod_view.aspx?TypeId=50067&Id=1272&FId=t3:50067:3

Which pins would you use here to debug with JTAG?

Jason2866, 2022-10-06:
@nest-don  Ahh, sorry it is only a ESP32 with that device, i had the one with the ESP32-S3 in mind. So the answer is indeed not possible!