Xinyuan-LilyGO/LilyGo-Camera-Series #8: How to upload sketch from Arduino IDE to T-SimCam

eloquentarduino, 2022-11-25:
I bought a T-SimCam. How do I upload a sketch from the Arduino IDE?
Simply hitting upload doesn't work.
Do I need to press RST or BOOT buttons in any given sequence?

lbuque, 2022-11-25:
The pc is directly connected to the t-simcam via usb.

rIf it cannot be downloaded automatically, you need to manually enter the download mode. (This operation is the same as other esp32 chips)

eloquentarduino, 2022-11-25:
Can you please describe step by step what I have to do?

lbuque, 2022-11-26:
The following configuration is recommended:

[image]

closed 2022-12-05 by lbuque

eloquentarduino, 2022-12-06:
I was able to upload a sketch only once. Now it throws error while uploading.

```
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:21:a6:5c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Error with exit status 1
```

abratchik, 2023-01-26:
Just released a new version of [esp32-web-server](https://github.com/abratchik/esp32-cam-webserver), you may test with this board. Working great for me so far, thanks a lot!

RogerParkinson, 2023-08-22:
Did this actually get resolved or just closed? Questions I have:

- Was there some sequence of buttons that needs pressing? 
- Did the problem with uploading the sketch which then threw errors get resolved?

RogerParkinson, 2023-08-22:
Okay for the next guy, I have it working now. I did not have to press any buttons. I just needed these settings
[image]
This is using the sketch at examples/sketch