Xinyuan-LilyGO/LilyGo-Camera-Series #14: T-Camera v162 on battery?

tomtobback, 2023-03-13:
I have a T-Camera v1.6.2 which works well with esphome when powered by USB. however, when i connect a Li battery on the PH connector, the 3V3 line does not get any voltage. i looked at the [schematic](https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series/blob/master/schematic/T_CameraV162_Schematic.pdf) and i cannot find any trace from VBAT to the DC/DC or LDO, how is this supposed to work?

JunliangYe, 2023-03-16:
I managed to boot up by pressing the rst button

tomtobback, 2023-03-20:
thanks @JunliangYe you're right, i guess i did not press hard enough.
it draws about 150mA running esphome, not very battery friendly. any advice on how to hack v1.6.2 to get the PIR connected to an external wake-up pin?

closed 2023-03-20 by tomtobback