Xinyuan-LilyGO/LilyGo-Camera-Series #16: Help : Connecting HX711

JpEncausse, 2023-03-31:
Hello,
Is it possible to connect an [HX711 module](https://www.diyengineers.com/2022/05/19/load-cell-with-hx711-how-to-use-with-examples/) to the camera board v 1.6.2 with JH2.0 port ?

[image]

I'm using the sketch template. Everything works fine (many thanks !) but when connecting the [HX711 module](https://www.diyengineers.com/2022/05/19/load-cell-with-hx711-how-to-use-with-examples/) I have the following error :

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008bb14  PS      : 0x00060930  A0      : 0x800d56a3  A1      : 0x3ffb2760  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000400  A5      : 0x00000000
A6      : 0x3ffb8a28  A7      : 0x00000040  A8      : 0x006ec580  A9      : 0x3ff5f078  
A10     : 0x3ff5f07c  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x3ffb8a28  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008bb14  LEND    : 0x4008bb1f  LCOUNT  : 0x0000003f  

Backtrace:0x4008bb11:0x3ffb27600x400d56a0:0x3ffb2770 0x400d5bc9:0x3ffb2790 0x400d5cb9:0x3ffb27b0 0x400d3c5a:0x3ffb27e0 0x400d46d7:0x3ffb2800 0x400ddd15:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
```

So the board reboot. 
- I tried to connect only GND and VCC but it keep rebooting
- I tried IO21 and IO22 on both ways to DC & SCK but it keep rebooting
- I comment `setupSensor()` and `readSensor()` assuming it has a side effect

According to the documentation HX711 is I2C compatible and the code compile correctly so I try to figure out what happened ? What should I do to make it work ? 

I understand the difference with BME280 that is using other port SDA and SCL but I'm not skilled enought to understand the wire-ing to do. If you can point me into the right direction ? Many Thanks !

lewisxhe, 2023-04-04:
I haven't used HX711, but I can see from the device specification that it does not belong to the I2C communication method

lewisxhe, 2023-04-04:
For re-information, you can install EspExceptionDecoder to decode and view the cause of the error 
https://github.com/me-no-dev/EspExceptionDecoder

JpEncausse, 2023-04-04:
> I haven't used HX711, but I can see from the device specification that it does not belong to the I2C communication method

OMG thanks I thought I had read it was. I'll order [the one from DFRobotics](https://www.dfrobot.com/product-2289.html)

> For re-information, you can install EspExceptionDecoder to decode and view the cause of the error https://github.com/me-no-dev/EspExceptionDecoder

Thanks ! I'll try to install it on PlatformIO

Do you recommend an I2C brand working well with the Camera ? (I'm looking for LED, NFC, Touch ...
Many thanks !

lewisxhe, 2023-04-04:
Doesn't the board already have an I2C OLED display?

JpEncausse, 2023-04-04:
Yes I think. I try to add modules to the board:
- https://fr.aliexpress.com/item/1005005059816321.html
- https://fr.aliexpress.com/item/32968683765.html

Using the I2C extension like it is done and explain with BME280 module.
But with other modules. I try to figureout what is compatible.

JpEncausse, 2023-04-05:
I try to figure out how to make it work PN532 in I2C but still no success
https://www.electroschematics.com/nfc-rfid-module-pn532/

I correclty wire the camera as schema explain
[image]

In my case I use 5V, GND, IO22 and IO21 and for the code I follow the sensor code:

```
#include <PN532_I2C.h>
#include <PN532.h> 
#include <NfcAdapter.h>

PN532_I2C pn532_i2c(Wire);
NfcAdapter nfc = NfcAdapter(pn532_i2c);

bool setupNFC(){
    Wire.begin(I2C_SDA, I2C_SCL);
    nfc.begin();
    return false;
}  

bool loopNFC(){
    delay(100);
    if (nfc.tagPresent()) {
        NfcTag tag = nfc.read();
        tag.print();
    }
}
```

But receive an error `Didn't find PN53x board` I assume I missed something I should do ?

lewisxhe, 2023-04-05:
There are no errors in the sketch, but there may be issues with the hardware connection. I remember that this module needs to be switched to I2C communication mode in order to be used

JpEncausse, 2023-04-05:
> There are no errors in the sketch, but there may be issues with the hardware connection. I remember that this module needs to be switched to I2C communication mode in order to be used

Yes I set the PIN to I2C,
I'll double check the module on an other board to check if it works

closed 2023-04-25 by lewisxhe