Xinyuan-LilyGO/LilyGo-Camera-Series #20: Unable to use TFT screen on `t-camera-plus`

duchengyao, 2023-06-23:
I use `example/sketch`, no screen output.

其它功能正常，只是屏幕有背光无内容，黑屏。

`TFT_eSPI @ 2.4.`**`78`** is not available, so I use `TFT_eSPI @ 2.4.`**`79`** in the platformio.ini below.

platformio.ini
```
;PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html
[platformio]
src_dir = .

[env]
platform = espressif32
board = esp32dev
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.3
framework = arduino

; Serial Monitor options
monitor_speed = 115200
; monitor_port = COM137
; upload_port = COM137

[env:t-camera-plus]
extends = env
board = esp32dev
framework = arduino
; upload_port = COM5
build_flags = -D BOARD_HAS_PSRAM
              -D T_Camera_PLUS_VERSION

lib_deps =
    ; ESP8266_SSD1306@4.1.0
    ; mathertel/OneButton @ 2.0.3
    bodmer/TFT_eSPI @ 2.4.79
    ; https://github.com/ThingPulse/esp8266-oled-ssd1306#4.3.0
    ; https://github.com/sparkfun/SparkFun_BME280_Arduino_Library#v2.0.10
```

console output:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
setupDisplay status 1
[  1234][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=400DC70C arg=3FFBDC64
[  1541][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  1541][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  1847][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
setupSDCard status 0
setupPower status 1
setupSensor status 1
setupCamera status 1
.
WiFi connected
Starting web server on port: '80'
Starting stream server on port: '81'
Camera Ready! Use 'http://192.168.2.195' to connect
```

lewisxhe, 2023-06-24:
选中对应驱动文件https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L78-L78

duchengyao, 2023-06-24:
感谢回复，我使用 platformio ，在 `.pio/libdeps/t-camera-plus/TFT-eSPI/User_Setup_Select.h` 里面取消注释了 `#include <User_Setups/Setup44_TTGO_CameraPlus.h>`，我是  2.4.79 所以是第81行

https://github.com/Bodmer/TFT_eSPI/blob/ef7c02098076c9a3819c30faf7da5d4af77db008/User_Setup_Select.h#L81

屏幕有显示了，但是显示有错误，我拍了视频，如下：

https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series/assets/7941237/dae3ef31-dba5-4d15-9e23-2e6a4f14a6b1

duchengyao, 2023-06-24:
另外，在编译的时候会报几个 redefined 的 warnning ，请问我需要进行哪些修改

```
In file included from .pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup_Select.h:81,
                 from .pio/libdeps/t-camera-plus/TFT_eSPI/TFT_eSPI.h:68,
                 from C:/Users/hbyls/Documents/Arduino/LilyGo-Camera-Series/examples/sketch/sketch.ino:82:
.pio/libdeps/t-camera-plus/TFT_eSPI/User_Setups/Setup44_TTGO_CameraPlus.h:11: warning: "TFT_CS" redefined
 #define TFT_CS      12

In file included from .pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup_Select.h:30,
                 from .pio/libdeps/t-camera-plus/TFT_eSPI/TFT_eSPI.h:68,
                 from C:/Users/hbyls/Documents/Arduino/LilyGo-Camera-Series/examples/sketch/sketch.ino:82:
.pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup.h:169: note: this is the location of the previous definition
 #define TFT_CS   PIN_D8  // Chip select control pin D8

In file included from .pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup_Select.h:81,
                 from .pio/libdeps/t-camera-plus/TFT_eSPI/TFT_eSPI.h:68,
                 from C:/Users/hbyls/Documents/Arduino/LilyGo-Camera-Series/examples/sketch/sketch.ino:82:
.pio/libdeps/t-camera-plus/TFT_eSPI/User_Setups/Setup44_TTGO_CameraPlus.h:12: warning: "TFT_DC" redefined
 #define TFT_DC      15

In file included from .pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup_Select.h:30,
                 from .pio/libdeps/t-camera-plus/TFT_eSPI/TFT_eSPI.h:68,
                 from C:/Users/hbyls/Documents/Arduino/LilyGo-Camera-Series/examples/sketch/sketch.ino:82:
.pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup.h:170: note: this is the location of the previous definition
 #define TFT_DC   PIN_D3  // Data Command control pin

In file included from .pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup_Select.h:81,
                 from .pio/libdeps/t-camera-plus/TFT_eSPI/TFT_eSPI.h:68,
                 from C:/Users/hbyls/Documents/Arduino/LilyGo-Camera-Series/examples/sketch/sketch.ino:82:
.pio/libdeps/t-camera-plus/TFT_eSPI/User_Setups/Setup44_TTGO_CameraPlus.h:13: warning: "TFT_RST" redefined
 #define TFT_RST     -1

In file included from .pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup_Select.h:30,
                 from .pio/libdeps/t-camera-plus/TFT_eSPI/TFT_eSPI.h:68,
                 from C:/Users/hbyls/Documents/Arduino/LilyGo-Camera-Series/examples/sketch/sketch.ino:82:
.pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup.h:171: note: this is the location of the previous definition
 #define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)

In file included from .pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup_Select.h:81,
                 from .pio/libdeps/t-camera-plus/TFT_eSPI/TFT_eSPI.h:68,
                 from C:/Users/hbyls/Documents/Arduino/LilyGo-Camera-Series/examples/sketch/sketch.ino:82:
.pio/libdeps/t-camera-plus/TFT_eSPI/User_Setups/Setup44_TTGO_CameraPlus.h:31: warning: "SPI_FREQUENCY" redefined
 #define SPI_FREQUENCY  40000000 // Maximum to use SPIFFS
 
In file included from .pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup_Select.h:30,
                 from .pio/libdeps/t-camera-plus/TFT_eSPI/TFT_eSPI.h:68,
                 from C:/Users/hbyls/Documents/Arduino/LilyGo-Camera-Series/examples/sketch/sketch.ino:82:
.pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup.h:343: note: this is the location of the previous definition
 #define SPI_FREQUENCY  27000000
 
In file included from .pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup_Select.h:189,
                 from .pio/libdeps/t-camera-plus/TFT_eSPI/TFT_eSPI.h:68,
                 from C:/Users/hbyls/Documents/Arduino/LilyGo-Camera-Series/examples/sketch/sketch.ino:82:
.pio/libdeps/t-camera-plus/TFT_eSPI/TFT_Drivers/ILI9341_Defines.h:6: warning: "TFT_HEIGHT" redefined
   #define TFT_HEIGHT 320

In file included from .pio/libdeps/t-camera-plus/TFT_eSPI/User_Setup_Select.h:81,
                 from .pio/libdeps/t-camera-plus/TFT_eSPI/TFT_eSPI.h:68,
                 from C:/Users/hbyls/Documents/Arduino/LilyGo-Camera-Series/examples/sketch/sketch.ino:82:
.pio/libdeps/t-camera-plus/TFT_eSPI/User_Setups/Setup44_TTGO_CameraPlus.h:6: note: this is the location of the previous definition
 #define TFT_HEIGHT 240

In file included from C:/Users/hbyls/Documents/Arduino/LilyGo-Camera-Series/examples/sketch/sketch.ino:82:
.pio/libdeps/t-camera-plus/TFT_eSPI/TFT_eSPI.h:909:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
        ^~~~~~~
```

duchengyao, 2023-06-24:
请问是否有可以同时开启 rtsp stream 和 tft 显示 的样例

lewisxhe, 2023-06-24:
注释默认pin配置 https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h

duchengyao, 2023-06-25:
这些吗？不起作用。

https://github.com/Bodmer/TFT_eSPI/blob/eb68e056d95f0f62bc08eab9242f5855518917cd/User_Setup_Select.h#L282-L300

duchengyao, 2023-06-25:
比如 SPI_FREQUENCY 到底应该是 40000000 还是 27000000

lewisxhe, 2023-06-25:
注释默认pin配置 https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h

27行

duchengyao, 2023-06-26:
我注释了27行和取消注释`SPI.begin(TFT_SCLK_PIN, TFT_MISO_PIN, TFT_MOSI_PIN);`以后，正常运行了。万分感谢。

请问我为什么要取消注释`SPI.begin(TFT_SCLK_PIN, TFT_MISO_PIN, TFT_MOSI_PIN);`？ 按注释写的，应该TFT已经初始化过了吧

closed 2023-06-28 by duchengyao