Xinyuan-LilyGO/LilyGo-Camera-Series #27: T-SIMCAM and battery not working

RogerParkinson, 2023-11-19:
I have a T-SIMCAM which is working just fine when I power it with the USB. It has a SIM7000E on it and that happily talks to my cell network. All good. It shows a blue light near the battery connector, a red status light on the modem and a flashing netlight also on the modem.

Now I want to power it from the battery. I plugged in a fully charged LiPo battery, nominally 3.7v but my multimeter shows 4.1v. Nothing happened. No lights, and the program I have flashed is clearly not running.
Is there something else I need to do to enable powering this device by battery?

RogerParkinson, 2023-11-28:
Still no answer here. I tried to ask on the LilyGo forum but all my attempts to register there give me a 504 error.
I've looked at the schematic and I can see the battery delivers 3.3v to the ESP32 (and I measured that my battery does that), but there's no evidence the program is running.

lewisxhe, 2023-11-28:
GPIO1 must be set to output during setup, and the output must be set to high level before the battery can be used.
https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series/blob/17ea98b9b37734d08c80924c1630afda42c5e66a/examples/t_sim_cam_factory/t_sim_cam_factory.ino#L26