Xinyuan-LilyGO/LilyGo-Camera-Series #29: simcam grove connection problem

martinbakker54, 2024-01-13:
Trying to use the grove connection with a PCF8574 i2c io expander.  Noticed that  the connector wiring is not compliant with Grove but solved this by using a led to find vcc and ground. Using i2c scanner software and looking at the electronic drawings  I noticed that  sda pin = 3 and scl pin = 5. The scanner finds 126 addresses. 

Now my questions is: what are the sda and scl pin numbers for the grove connection.

lewisxhe, 2024-03-05:
I have already given feedback on the problem with the drawings
[image]

anthonyhoegberg, 2024-06-07:
@lewisxhe 

With the new board of V1.3 Of T-SIMCAM
I see there is still a problem with non compliant grove connector, however what GPIO pins would be correct for the data pins in the middle, GPIO43, GPIO44 is this still correct?