Xinyuan-LilyGO/LilyGo-Camera-Series #30: T-SIMCAM SD card read/write problem

EvanChristan123, 2024-01-18:
I am doing the project on the T-SIMCAM board and trying to make the board recognize MicroSD card. The MicroSD is formatted into exFAT and uses the example/sketch to test. Unfortunately, the program didn't recognize the MicroSD inserted before the board activated even though the other parts of the program were functioning. I also have tried to redefine the SDCARD_CS_PIN in  select_pins.h as follows:
```
.
.
.
#elif defined(CAMERA_MODEL_TTGO_T_CAM_SIM)
#define PWDN_GPIO_NUM  -1
#define RESET_GPIO_NUM 18
#define XCLK_GPIO_NUM  14
#define SIOD_GPIO_NUM  4
#define SIOC_GPIO_NUM  5

#define Y9_GPIO_NUM    15
#define Y8_GPIO_NUM    16
#define Y7_GPIO_NUM    17
#define Y6_GPIO_NUM    12
#define Y5_GPIO_NUM    10
#define Y4_GPIO_NUM    8
#define Y3_GPIO_NUM    9
#define Y2_GPIO_NUM    11
#define VSYNC_GPIO_NUM 6
#define HREF_GPIO_NUM  7
#define PCLK_GPIO_NUM  13

#define SDCARD_CS_PIN   47

#define PWR_ON_PIN     1
#define PCIE_PWR_PIN   48
#define PCIE_RST_PIN   48
#define PCIE_TX_PIN    45
#define PCIE_RX_PIN    46
#define PCIE_LED_PIN   21
#else
.
.
.
```
the program still couldn't do anything related to the SD card. Can somebody help me explain how the T-SIMCAM SD card read/write works, please?

martinbakker54, 2024-01-18:
I have the following lines in my code for the sd card pins. _

//define sd card
#include "FS.h"  
#include "SD.h"
     
#define SD_MISO_PIN      40
#define SD_MOSI_PIN      38
#define SD_SCLK_PIN      39
#define SD_CS_PIN        47

EvanChristan123, 2024-01-18:
We actually set SDCARD_CS_PIN to 47 and we assume the board will read the info of the card through this port, but the board just can't read the card. Since the example program doesn't give any hints on using the other three ports, may I ask how to use these 4 pins to operate the SD card with "FS.h" and "SD.h"?

martinbakker54, 2024-01-18:
I use this code:

SPI.begin(SD_SCLK_PIN, SD_MISO_PIN, SD_MOSI_PIN, SD_CS_PIN);
    if (!SD.begin(SD_CS_PIN, SPI)) {
        Serial.println("Card Mount Failed");
        return;
    }
    uint8_t cardType = SD.cardType();
    if (cardType == CARD_NONE) {
        Serial.println("No SD card attached");
        return;
    }

    Serial.print("SD Card Type: ");

    if (cardType == CARD_MMC)
        Serial.println("MMC");
    else if (cardType == CARD_SD)
        Serial.println("SDSC");
    else if (cardType == CARD_SDHC)
        Serial.println("SDHC");
    else
        Serial.println("UNKNOWN");

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);
//    SD.end();

EvanChristan123, 2024-01-19:
It works! When I also changed the platform.ini to include the TFT_eSPI instead of ignoring it, the SD can be read perfectly. Thank you so much for your code.