Xinyuan-LilyGO/LilyGo-Camera-Series #31: T_CameraMini doesn't works on ESPHome with provided yaml : ESP_ERR_NOT_FOUND

xbeaudouin, 2024-03-01:
Hello,

I used the given Yaml to program with ESPHome a T_CameraMini but the camera is not found.
Checked the source code and the yaml but no luck.
Is there a specific hardware that has been changed ?

Errors:
```
[12:12:39][I][app:102]: ESPHome version 2024.2.1 compiled on Mar  1 2024, 12:09:28
[12:12:39][C][wifi:577]: WiFi:
[12:12:39][C][wifi:409]:   Local MAC: 08:3A:F2:XX:XX:XX
[12:12:39][C][wifi:414]:   SSID: [redacted]
[12:12:39][C][wifi:415]:   IP Address: 192.168.xx.xx
[12:12:39][C][wifi:417]:   BSSID: [redacted]
[12:12:39][C][wifi:418]:   Hostname: 'espcam1'
[12:12:39][C][wifi:420]:   Signal strength: -44 dB ▂▄▆█
[12:12:39][C][wifi:424]:   Channel: 1
[12:12:39][C][wifi:425]:   Subnet: 255.255.255.0
[12:12:39][C][wifi:426]:   Gateway: 192.168.xx.xx
[12:12:39][C][wifi:427]:   DNS1: 192.168.xx.xx
[12:12:39][C][wifi:428]:   DNS2: 192.168.xx.xx
[12:12:39][C][logger:447]: Logger:
[12:12:39][C][logger:448]:   Level: DEBUG
[12:12:39][C][logger:449]:   Log Baud Rate: 115200
[12:12:39][C][logger:451]:   Hardware UART: UART0
[12:12:39][C][i2c.arduino:053]: I2C Bus:
[12:12:39][C][i2c.arduino:054]:   SDA Pin: GPIO21
[12:12:39][C][i2c.arduino:055]:   SCL Pin: GPIO22
[12:12:39][C][i2c.arduino:056]:   Frequency: 50000 Hz
[12:12:39][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[12:12:39][I][i2c.arduino:069]: Results from i2c bus scan:
[12:12:39][I][i2c.arduino:075]: Found i2c device at address 0x34
[12:12:39][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Test Button'
[12:12:39][C][gpio.binary_sensor:016]:   Pin: GPIO33
[12:12:39][C][uptime.sensor:031]: Uptime Sensor 'Test Uptime'
[12:12:39][C][uptime.sensor:031]:   Device Class: 'duration'
[12:12:39][C][uptime.sensor:031]:   State Class: 'total_increasing'
[12:12:39][C][uptime.sensor:031]:   Unit of Measurement: 's'
[12:12:39][C][uptime.sensor:031]:   Accuracy Decimals: 0
[12:12:39][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[12:12:39][C][status:034]: Status Binary Sensor 'Test Status'
[12:12:39][C][status:034]:   Device Class: 'connectivity'
[12:12:39][C][esp32_camera:048]: ESP32 Camera:
[12:12:39][C][esp32_camera:049]:   Name: Test Camera
[12:12:39][C][esp32_camera:050]:   Internal: NO
[12:12:39][C][esp32_camera:052]:   Data Pins: D0:5 D1:14 D2:4 D3:15 D4:37 D5:38 D6:36 D7:39
[12:12:39][C][esp32_camera:053]:   VSYNC Pin: 27
[12:12:39][C][esp32_camera:054]:   HREF Pin: 25
[12:12:39][C][esp32_camera:055]:   Pixel Clock Pin: 19
[12:12:39][C][esp32_camera:056]:   External Clock: Pin:32 Frequency:20000000
[12:12:39][C][esp32_camera:060]:   I2C Pins: SDA:13 SCL:12
[12:12:39][C][esp32_camera:062]:   Reset Pin: -1
[12:12:39][C][esp32_camera:080]:   Resolution: 640x480 (VGA)
[12:12:39][E][esp32_camera:123]:   Setup Failed: ESP_ERR_NOT_FOUND
[12:12:39][C][homeassistant.time:010]: Home Assistant Time:
[12:12:39][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[12:12:39][C][psram:020]: PSRAM:
[12:12:39][C][psram:021]:   Available: YES
[12:12:39][C][psram:024]:   Size: 4095 KB
[12:12:40][C][captive_portal:088]: Captive Portal:
[12:12:40][C][mdns:115]: mDNS:
[12:12:40][C][mdns:116]:   Hostname: espcam1
[12:12:40][C][ota:096]: Over-The-Air Updates:
[12:12:40][C][ota:097]:   Address: espcam1.local:3232
[12:12:40][C][ota:100]:   Using Password.
[12:12:40][C][ota:103]:   OTA version: 2.
[12:12:40][D][api:102]: Accepted 192.168.xx.xx
[12:12:40][C][api:139]: API Server:
[12:12:40][C][api:140]:   Address: espcam1.local:6053
[12:12:40][C][api:142]:   Using noise encryption: YES
[12:12:40][C][wifi_signal.sensor:009]: WiFi Signal 'Test WiFi Signal'
[12:12:40][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[12:12:40][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[12:12:40][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[12:12:40][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
```

Here yaml I used:

```
esphome:
  name: espcam1
  friendly_name: espcam1

substitutions:
  friendly_name: Test

esp32:
  board: esp32dev

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "<censored>"

ota:
  password: "<censored>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Espcam1 Fallback Hotspot"
    password: "censored"

captive_portal:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO33
      mode: INPUT_PULLUP
      inverted: True
    name: $friendly_name Button
    filters:
      - delayed_off: 50ms
  - platform: status
    name: $friendly_name Status

sensor:
  - platform: wifi_signal
    name: $friendly_name WiFi Signal
    update_interval: 10s
  - platform: uptime
    name: $friendly_name Uptime

esp32_camera:
  name: $friendly_name Camera
  external_clock:
    pin: GPIO32
    frequency: 20MHz
  i2c_pins:
    sda: GPIO13
    scl: GPIO12
  data_pins: [GPIO5, GPIO14, GPIO4, GPIO15, GPIO37, GPIO38, GPIO36, GPIO39]
  vsync_pin: GPIO27
  href_pin: GPIO25
  pixel_clock_pin: GPIO19
  #power_down_pin: GPIO26
  #resolution: 640x480
  #jpeg_quality: 10
  #vertical_flip: true
  #horizontal_mirror: false

i2c:
  sda: GPIO21
  scl: GPIO22

time:
  - platform: homeassistant
    id: homeassistant_time
```

How can I find the right GPIO for this cam ?

lewisxhe, 2024-03-05:
CameraMini has a PMU. You must turn on the PMU power before use.

xbeaudouin, 2024-03-06:
Well I slightly modified the ESPhome yaml

```
esphome:
  name: espcam1
  friendly_name: espcam1

substitutions:
  friendly_name: Test

esp32:
  board: esp32dev

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "ffAjE/GBPOrV2bIDdMWkszndD3WgbyJOAQBTHddx9kc="

ota:
  password: "954247cc9c33d293ddd20f7811d898bd"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Espcam1 Fallback Hotspot"
    password: "1ZC4GGHLjp8z"

captive_portal:

external_components:
  - source: github://pionizer/pionizer-axp192
    components: [axp192]

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO33
      mode: INPUT_PULLUP
      inverted: False
    name: $friendly_name Button
    filters:
      - delayed_off: 50ms
  - platform: status
    name: $friendly_name Status

sensor:
  - platform: wifi_signal
    name: $friendly_name WiFi Signal
    update_interval: 10s
  - platform: uptime
    name: $friendly_name Uptime
  - platform: axp192
    model: m5stickc
    id: power_mgmt
    address: 0x34
    #i2c_id: i2c_bus
    update_interval: 60s
    brightness: 100%
    battery_level:
      name: $friendly_name Battery Level
      id: batterylevel

esp32_camera:
  name: $friendly_name Camera
  external_clock:
    pin: GPIO32
    frequency: 20MHz
  i2c_pins:
    sda: GPIO13
    scl: GPIO12
  data_pins: [GPIO5, GPIO14, GPIO4, GPIO15, GPIO37, GPIO38, GPIO36, GPIO39]
  vsync_pin: GPIO27
  href_pin: GPIO25
  pixel_clock_pin: GPIO19
  #power_down_pin: GPIO26
  #resolution: 640x480
  #jpeg_quality: 10
  #vertical_flip: true
  #horizontal_mirror: false

i2c:
  sda: GPIO21
  scl: GPIO22

time:
  - platform: homeassistant
    id: homeassistant_time
```

As you see there now an apx192 at address 0x34, and... it is detected, but doesn't seems to change anything (on ESPhome 2024.2.2):

```
[11:52:31][C][status:034]:   Device Class: 'connectivity'
[11:52:31][C][axp192.sensor:047]: AXP192:
[11:52:31][C][axp192.sensor:048]:   Address: 0x34
[11:52:31][C][axp192.sensor:049]:   Battery Level 'Test Battery Level'
[11:52:31][C][axp192.sensor:049]:     State Class: ''
[11:52:31][C][axp192.sensor:049]:     Unit of Measurement: '%'
[11:52:31][C][axp192.sensor:049]:     Accuracy Decimals: 1
[11:52:31][C][axp192.sensor:049]:     Icon: 'mdi:battery'
[11:52:31][C][esp32_camera:048]: ESP32 Camera:
[11:52:31][C][esp32_camera:049]:   Name: Test Camera
[11:52:31][C][esp32_camera:050]:   Internal: NO
[11:52:31][C][esp32_camera:052]:   Data Pins: D0:5 D1:14 D2:4 D3:15 D4:37 D5:38 D6:36 D7:39
[11:52:31][C][esp32_camera:053]:   VSYNC Pin: 27
[11:52:31][C][esp32_camera:054]:   HREF Pin: 25
[11:52:31][C][esp32_camera:055]:   Pixel Clock Pin: 19
[11:52:31][C][esp32_camera:056]:   External Clock: Pin:32 Frequency:20000000
[11:52:31][C][esp32_camera:060]:   I2C Pins: SDA:13 SCL:12
[11:52:31][C][esp32_camera:062]:   Reset Pin: -1
[11:52:31][C][esp32_camera:080]:   Resolution: 640x480 (VGA)
[11:52:31][E][esp32_camera:123]:   Setup Failed: ESP_ERR_NOT_FOUND
[11:52:31][C][homeassistant.time:010]: Home Assistant Time:
[11:52:31][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
```

Maybe there is something missing somewhere?

Kind regards

xbeaudouin, 2024-03-07:
Ok I get it detected... But still have some issues, I have added the following to esphome yaml:
```
## To enable camera...
status_led:
  pin: 
    number: 26
    inverted: True

esp32_camera_web_server:
  - port: 8080
    mode: stream
  - port: 8081
    mode: snapshot
```

But I got now `[W][esp32_camera:190]: Got invalid frame from camera!`, can you give me some more informations?

```
[11:12:55][C][i2c.arduino:053]: I2C Bus:
[11:12:55][C][i2c.arduino:054]:   SDA Pin: GPIO21
[11:12:55][C][i2c.arduino:055]:   SCL Pin: GPIO22
[11:12:55][C][i2c.arduino:056]:   Frequency: 50000 Hz
[11:12:55][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[11:12:55][I][i2c.arduino:069]: Results from i2c bus scan:
[11:12:55][I][i2c.arduino:075]: Found i2c device at address 0x34
[11:12:55][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Test Button'
[11:12:55][C][gpio.binary_sensor:016]:   Pin: GPIO33
[11:12:55][C][uptime.sensor:031]: Uptime Sensor 'Test Uptime'
[11:12:55][C][uptime.sensor:031]:   Device Class: 'duration'
[11:12:55][C][uptime.sensor:031]:   State Class: 'total_increasing'
[11:12:55][C][uptime.sensor:031]:   Unit of Measurement: 's'
[11:12:55][C][uptime.sensor:031]:   Accuracy Decimals: 0
[11:12:55][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[11:12:55][C][status:034]: Status Binary Sensor 'Test Status'
[11:12:55][C][status:034]:   Device Class: 'connectivity'
[11:12:55][C][axp192.sensor:047]: AXP192:
[11:12:55][C][axp192.sensor:048]:   Address: 0x34
[11:12:55][C][axp192.sensor:049]:   Battery Level 'Test Battery Level'
[11:12:55][C][axp192.sensor:049]:     State Class: ''
[11:12:55][C][axp192.sensor:049]:     Unit of Measurement: '%'
[11:12:55][C][axp192.sensor:049]:     Accuracy Decimals: 1
[11:12:55][C][axp192.sensor:049]:     Icon: 'mdi:battery'
[11:12:55][C][esp32_camera:048]: ESP32 Camera:
[11:12:55][C][esp32_camera:049]:   Name: Test Camera
[11:12:55][C][esp32_camera:050]:   Internal: NO
[11:12:55][C][esp32_camera:052]:   Data Pins: D0:5 D1:14 D2:4 D3:15 D4:37 D5:38 D6:36 D7:39
[11:12:55][C][esp32_camera:053]:   VSYNC Pin: 27
[11:12:55][C][esp32_camera:054]:   HREF Pin: 25
[11:12:55][C][esp32_camera:055]:   Pixel Clock Pin: 19
[11:12:55][C][esp32_camera:056]:   External Clock: Pin:32 Frequency:20000000
[11:12:55][C][esp32_camera:060]:   I2C Pins: SDA:13 SCL:12
[11:12:55][C][esp32_camera:062]:   Reset Pin: -1
[11:12:55][C][esp32_camera:080]:   Resolution: 640x480 (VGA)
[11:12:55][C][esp32_camera:129]:   JPEG Quality: 10
[11:12:55][C][esp32_camera:131]:   Contrast: 0
[11:12:55][C][esp32_camera:132]:   Brightness: 0
[11:12:55][C][esp32_camera:133]:   Saturation: 0
[11:12:55][C][esp32_camera:134]:   Vertical Flip: ON
[11:12:55][C][esp32_camera:135]:   Horizontal Mirror: ON
[11:12:55][C][esp32_camera:136]:   Special Effect: 0
[11:12:55][C][esp32_camera:137]:   White Balance Mode: 0
[11:12:55][C][esp32_camera:140]:   Auto Exposure Control: 1
[11:12:55][C][esp32_camera:141]:   Auto Exposure Control 2: 0
[11:12:55][C][esp32_camera:142]:   Auto Exposure Level: 0
[11:12:55][C][esp32_camera:143]:   Auto Exposure Value: 300
[11:12:55][C][esp32_camera:144]:   AGC: 1
[11:12:55][C][esp32_camera:145]:   AGC Gain: 0
[11:12:55][C][esp32_camera:146]:   Gain Ceiling: 0
[11:12:55][C][esp32_camera:152]:   Test Pattern: NO
[11:12:55][C][homeassistant.time:010]: Home Assistant Time:
[11:12:55][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[11:12:55][C][psram:020]: PSRAM:
[11:12:55][C][psram:021]:   Available: YES
[11:12:55][C][psram:024]:   Size: 4095 KB
[11:12:55][C][captive_portal:088]: Captive Portal:
[11:12:55][C][web_server:168]: Web Server:
[11:12:55][C][web_server:169]:   Address: espcam1.local:80
[11:12:55][C][mdns:115]: mDNS:
[11:12:55][C][mdns:116]:   Hostname: espcam1
[11:12:55][C][ota:096]: Over-The-Air Updates:
[11:12:55][C][ota:097]:   Address: espcam1.local:3232
[11:12:55][C][ota:100]:   Using Password.
[11:12:55][C][ota:103]:   OTA version: 2.
[11:12:55][C][api:139]: API Server:
[11:12:55][C][api:140]:   Address: espcam1.local:6053
[11:12:55][C][api:142]:   Using noise encryption: YES
[11:12:55][C][wifi_signal.sensor:009]: WiFi Signal 'Test WiFi Signal'
[11:12:55][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[11:12:55][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[11:12:55][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[11:12:55][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[11:12:55][C][esp32_camera_web_server:088]: ESP32 Camera Web Server:
[11:12:55][C][esp32_camera_web_server:089]:   Port: 8080
[11:12:55][C][esp32_camera_web_server:091]:   Mode: stream
[11:12:55][C][esp32_camera_web_server:088]: ESP32 Camera Web Server:
[11:12:55][C][esp32_camera_web_server:089]:   Port: 8081
[11:12:55][C][esp32_camera_web_server:093]:   Mode: snapshot
[11:12:58][W][esp32_camera:190]: Got invalid frame from camera!
[11:12:58][D][sensor:094]: 'Test WiFi Signal': Sending state -54.00000 dBm with 0 decimals of accuracy
[11:13:02][W][esp32_camera:190]: Got invalid frame from camera!
```