Xinyuan-LilyGO/LilyGo-Camera-Series #32: SIMCOM SIM7600G-H PCIE module compatibility for LILYGO-T-SIMCAM

ojas-savarkar, 2024-03-04:
I have this [LILYGO-T-SIMCAM](https://www.lilygo.cc/products/t-simcam) Board 
On the website it is recommended to use the PCIE SIMCOM module made by them, in my case, [SIM7600G-H[H441-08]](https://www.lilygo.cc/products/a-t-pcie?variant=42335922028725).
However, it is not available to buy feasibly in my region, therefore, can I use this [waveshare SIM7600G-H-PCIE](https://www.waveshare.com/wiki/SIM7600G-H-PCIE) module instead?

Thanks in advance!

lewisxhe, 2024-03-05:
LilyGo's PCIE boards are all powered by 4V. You need to check whether other PCIE Pins and voltages are the same.

Jason2866, 2024-04-28:
Bought an mPCIe module from simcom. Works.