Xinyuan-LilyGO/LilyGo-Camera-Series #39: T-SIMCAM Groove interface

EvanChristan123, 2024-04-21:
Hi there, I have a few questions related to the groove on T-SIMCAM:

1) Can I use groove for LED light on T-SIMCAM?
2) If yes, is it compactable with seeed studio grove LED products? If yes , how can I work with these products? Do I need any converter with that?
3) Is there any example of using the groove interface on T-SIMCAM?

Hope somebody can help with that, thanks!

anthonyhoegberg, 2024-06-06:
Did u ever get an answer to this?

EvanChristan123, 2024-06-18:
Nope

EvanChristan123, 2024-06-18:
Nevertheless, base on #33, with 4 pin Female Jumper to Grove 4 pin Conversion Cable (https://www.seeedstudio.com/Grove-4-pin-Female-Jumper-to-Grove-4-pin-Conversion-Cable-5-PCs-per-PAck.html?p=2#detail-target-2) help, we can use Groove interface as addition gpio ports to control non-grove module, maximum 2 gpio ports provided I think

anthonyhoegberg, 2024-06-18:
https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series/issues/29#issuecomment-2154211026
if u look at this issue we can see the groove connector has an issue and is not compliant with groove standard so that needs to be fixed as well, and the pin mappings written in there seems to be correct

EvanChristan123, 2024-06-18:
That answered my second question

lewisxhe, 2024-07-26:
The groove interface can be configured as serial or GPIO or I2C. 
You only need to pass in the specified Pin during initialization.
 For example, 
I2C 
```
Wire.begin(43//SDA,44//SCL); 
```
Serial 
```
Serial2.begin(115200,SERIAL_8N1,43//RX,44//TX)
```