Xinyuan-LilyGO/LilyGo-Camera-Series #42: T-SIMCAM  V1.3 power/battery power spec?

anthonyhoegberg, 2024-06-05:
My company is currently doing testing with T-SIMCAM and might move to using it in production soon.
I cant really find documentation on voltage needed to power to T-SIMCAM V1.3 from non usb power source

I can see that we have a battery connector, and im wondering how that works, what mAh battery can it take spec wise, can it charge the battery or not etc

I would like max spec/recommended spec of battery, 
Direct pin power, non battery connector power on possible?
How to power the board without USB attach, like cold start from no power if battery only is available or non battery power method?
Can data about the battery be gathererd using the T-SIMCAM from the battery pin or is it power only?

anthonyhoegberg, 2024-06-07:
When switching from usb to battery power (cutting the power of the usb) the SIM modems power gets cut off, even tho PWR_EN pin is set to high

How can we fix this?

RogerParkinson, 2024-06-08:
I have a similar question, specifically about the max voltage for the battery connector. I'd like to test with my bench supply and I would like to not blow up the SIMCAM. Looking at the schematic there is a TP4056B connecting the VBUS (from USB) to the battery at P2 which suggests there is a charging function. I have not tried it though.
The VV_BAT from the battery is fed into a voltage regulator LN2220 to output VVBAT_4V2 which powers the modem, it also goes to a  SY8089AAC to output VDD3V3 which must be 3.3v for the ESP32. The [spec for the SY8089AAC](https://www.lcsc.com/datasheet/lcsc_datasheet_2108261630_Silergy-Corp-SY8089AAAC_C78988.pdf) says it accepts 2.7-5.5v but I cannot find a spec for the LN2220 in English. [This one](https://www.lcsc.com/datasheet/lcsc_datasheet_2105242005_NATLINEAR-LN2220PAR_C2833712.pdf) suggests it is okay for up to 24V but I can't read Chinese so I'm not sure.

lewisxhe, 2024-07-26:
> When switching from usb to battery power (cutting the power of the usb) the SIM modems power gets cut off, even tho PWR_EN pin is set to high
> 
> How can we fix this?

T-SIMCAM simply uses MOS tubes for power switching and cannot achieve seamless switching.

T-SIMCAM supports charging function. From the schematic diagram, it can be seen that TP4056 is used as charging management, RSET is set to 2K, that is, the charging current is set to about 580mA, and the standard 4.2V voltage is used as the charging cut-off voltage, so the battery that can be used is the standard 4.2V lithium-ion battery or 18650 battery

RogerParkinson, 2024-07-29:
Thanks @lewisxhe that answers my question.

RogerParkinson, 2024-07-31:
I have found that an old LIPO (nominally 3.7v) plugged into the battery terminal works when fully charged and I measure 4.12v from it.
I have a 18650 also nominally 3.7v which I measure 3.98v, but that fails to run the T-SIMCAM. I can't get diagnostics out of it without plugging in the USB which would deliver power too & defeat the purpose, but the modem lights flash in the same way as when it works on USB or the LIPO. That might mean the modem doesn't have enough power to connect or that the ESP32 is not getting enough power to ask it to connect. Either way it seems to need more than 3.98v.
It looks like these batteries will only work when really fully charged rather than just delivering their nominal voltage.

pol99b, 2024-08-29:
> I have found that an old LIPO (nominally 3.7v) plugged into the battery terminal works when fully charged and I measure 4.12v from it. I have a 18650 also nominally 3.7v which I measure 3.98v, but [...]

How did you solve it? Did you buy another type of battery?

RogerParkinson, 2024-08-30:
I found the 18650 does work if I fully charge it. I have since added three more 18650s in parallel (with fuses in between to prevent runaway discharges) and I figure this will hold the charge for longer. I do need it to last a while.

pol99b, 2024-08-30:
> I found the 18650 does work if I fully charge it. I have since added three more 18650s in parallel (with fuses in between to prevent runaway discharges) and I figure this will hold the charge for longer. I do need it to last a while.

How do you know when the 18650 is fully charge from the T-SIMCAM? Because my 18650 battery provides 3.7V, and when I connect the USB-C to charge it through the T-SIMCAM, it shows the blue LED when it should show red LED until it's fully charge with 4.2V. 

Also the blue LED it only turns on when I connect the T-SIMCAM with the USB-C, if I connect the T-SIMCAM to a battery or table power supply with the JST connector the LED it does not turn on but the T-SIMCAM works. (???)

RogerParkinson, 2024-09-02:
I have a separate charger so I haven't made much use of charging through the T-SIMCAM, sorry.
Like you I see the blue LED when I plug in the USB and no LED when I run off battery. I think that's correct behaviour.
I've never seen the red LED light up.