Xinyuan-LilyGO/LilyGo-Camera-Series #44: T-SIMCAM Esphome

illuzn, 2024-07-26:
I'm having trouble with the T-SIMCAM esphome yaml provided.

If trying to build using the latest 2024.7.2 I get a boot loop.

If trying to build using 2022.6.3 (when you produced that code), I get "KeyError: 'target_platform'"

This is the yaml I'm using for 2024.7.2
```yaml
# Doorbell configuration yaml

esphome:
  name: doorbell

esp32:
  board: esp32s3box
  variant: esp32s3
  flash_size: 16MB
  framework:
# There are build issues with esp32_camera and esp-idf as at July 2024. This is above my paygrade; use arduino for now.
    type: arduino

psram:
  mode: octal
  speed: 80MHz

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "doorbell_fallback"
    password: !secret fallback_password

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: !secret doorbell_api_key

ota:
  - platform: esphome
    password: !secret doorbell_ota_password

# T-SIMCAM Setup
esp32_camera:
  external_clock:
    pin: GPIO14
    frequency: 20MHz
  i2c_pins:
    sda: GPIO4
    scl: GPIO5
  data_pins: [GPIO11, GPIO9, GPIO8, GPIO10, GPIO12, GPIO17, GPIO16, GPIO15]
  vsync_pin: GPIO6
  href_pin: GPIO7
  pixel_clock_pin: GPIO13
  name: Camera
```

illuzn, 2024-07-26:
Also, which pins is the microphone connected to?

The schematic suggests GPIO2 (SDATA), GPIO47 (CLK) and GPIO48 (WS).
[image]

The product listing suggests GPIO2 (SDATA), GPIO41 (CLK) and GPIO42 (WS) - GPIO41 and GPIO42 are not connected according to the schematic.
[image]

lewisxhe, 2024-09-05:
Here you can see all the used GPIO
https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series/blob/master/examples/t_sim_cam_factory/config.h