Xinyuan-LilyGO/LilyGo-Camera-Series #45: T-SIMCAM with A7670E with GPS [H542] unable to get '+CGNSSPWR: READY!'

Daniel-De-Dev, 2024-08-05:
I've attempted to simply activate GNSS and retrieve location data as a simple test, yet even when i power and initialize A7670E it seems to never return `+CGNSSPWR: READY!` when sending `AT+CGNSSPWR=1` or `AT+CGNSSPWR=1,1,1`

I have attached the GPS antenna (not fully sure which port is for GPS but tried both). Further i have a placed the device outside with a clear sky and it just simply never manages to return the ready message. 

I would greatly appreciate some assistance, the code might be very bad and not follow normal conventions since this is my first experience with programming microcontrollers. The main goal was just to get the 'Ready!' response from the module and from there add logic to ask for location data, but it never does. I've had this running for more than an hour outside, more than enough time to get the data it needs but something is wrong, any ideas?

The code: 
```c
#include "driver/gpio.h"
#include "driver/uart.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include <string.h>

#define PCIE_RST_PIN    48
#define PCIE_TX_PIN     45
#define PCIE_RX_PIN     46
#define PCIE_LED_PIN    21

#define BUF_SIZE 1024

const uart_port_t UART_NUM_AT = UART_NUM_1;
static const char *TAG = "SIM_MODULE";

void send_at_command(const char* command);
bool read_response(QueueHandle_t uart_at_queue, const char* expected_response);

QueueHandle_t init() {
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
    };

    ESP_ERROR_CHECK(uart_param_config(UART_NUM_AT, &uart_config));
    ESP_ERROR_CHECK(uart_set_pin(UART_NUM_AT, PCIE_TX_PIN, PCIE_RX_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));

    QueueHandle_t uart_queue;
    ESP_ERROR_CHECK(uart_driver_install(UART_NUM_AT, BUF_SIZE * 2, 0, 10, &uart_queue, 0));

    gpio_config_t io_conf = {
        .intr_type = GPIO_INTR_DISABLE,
        .mode = GPIO_MODE_OUTPUT,
        .pin_bit_mask = (1ULL << PCIE_RST_PIN),
        .pull_down_en = 0,
        .pull_up_en = 0
    };
    gpio_config(&io_conf);
    
    // Power on AT module if it doesn't return OK
    send_at_command("AT");
    if (!read_response(uart_queue, "OK")) {
        ESP_LOGI(TAG, "Powering on AT module");
        gpio_set_level(PCIE_RST_PIN, 1);
        vTaskDelay(pdMS_TO_TICKS(50));
        gpio_set_level(PCIE_RST_PIN, 0);
        vTaskDelay(pdMS_TO_TICKS(50));
        gpio_set_level(PCIE_RST_PIN, 1);
        vTaskDelay(pdMS_TO_TICKS(11200)); // Wait for the module to initialize
    }

    ESP_LOGI(TAG, "PCI module initialized");
    return uart_queue;
}

void send_at_command(const char* command) {
    uart_write_bytes(UART_NUM_AT, command, strlen(command));
    uart_write_bytes(UART_NUM_AT, "\r\n", 2);
}

bool read_response(QueueHandle_t uart_at_queue, const char* expected_response) {
    uart_event_t event;
    char* data = (char*) malloc(BUF_SIZE);
    if (!data) {
        ESP_LOGE(TAG, "Failed to allocate memory for UART response");
        return false;
    }

    if (xQueueReceive(uart_at_queue, (void *)&event, pdMS_TO_TICKS(9000))) {
        memset(data, 0, BUF_SIZE);
        switch (event.type) {
            case UART_DATA: {
                int len = uart_read_bytes(UART_NUM_AT, data, BUF_SIZE - 1, pdMS_TO_TICKS(100));
                if (len > 0) {
                    data[len] = 0; // Null-terminate the string
                    ESP_LOGI(TAG, "Received data: %s", data);
                    if (expected_response && strstr(data, expected_response) != NULL) {
                        ESP_LOGI(TAG, "Got expected return value");
                        free(data);
                        return true;
                    }
                } else {
                    ESP_LOGW(TAG, "No data read from UART.");
                }
                break;
            }
            default:
                ESP_LOGW(TAG, "Unhandled UART event type: %d", event.type);
                break;
        }
    } else {
        ESP_LOGW(TAG, "No response within timeout period.");
    }

    free(data);
    return false;
}

void app_main(void) {
    QueueHandle_t uart_at_queue = init();

    send_at_command("AT+CGNSSPWR=1,1,1");

    while (true) {
        if (read_response(uart_at_queue, "+CGNSSPWR: READY!")) {
            ESP_LOGI(TAG, "GNSS module is ready.");
        }
        vTaskDelay(pdMS_TO_TICKS(5000)); // Add delay to avoid busy-waiting
    }
}
```

The console:
```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (26) boot: ESP-IDF v5.3 2nd stage bootloader
I (27) boot: compile time Aug  5 2024 11:04:15
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0b780h ( 46976) map
I (107) esp_image: segment 1: paddr=0001b7a8 vaddr=3fc93400 size=02a44h ( 10820) load
I (110) esp_image: segment 2: paddr=0001e1f4 vaddr=40374000 size=01e24h (  7716) load
I (118) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1aef8h (110328) map
I (144) esp_image: segment 4: paddr=0003af20 vaddr=40375e24 size=0d53ch ( 54588) load
I (163) boot: Loaded app from partition at offset 0x10000
I (163) boot: Disabling RNG early entropy source...
I (175) cpu_start: Multicore app
I (184) cpu_start: Pro cpu start user code
I (184) cpu_start: cpu freq: 160000000 Hz
I (184) app_init: Application information:
I (187) app_init: Project name:     test_gps
I (192) app_init: App version:      1
I (196) app_init: Compile time:     Aug  5 2024 11:03:51
I (202) app_init: ELF file SHA256:  9a3bb6930...
I (207) app_init: ESP-IDF:          v5.3
I (212) efuse_init: Min chip rev:     v0.0
I (217) efuse_init: Max chip rev:     v0.99 
I (222) efuse_init: Chip rev:         v0.2
I (227) heap_init: Initializing. RAM available for dynamic allocation:
I (234) heap_init: At 3FC96708 len 00053008 (332 KiB): RAM
I (240) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (246) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (252) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (259) spi_flash: detected chip: winbond
I (263) spi_flash: flash io: dio
W (267) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (280) sleep: Configure to isolate all GPIO pins in sleep state
I (295) main_task: Started on CPU0
I (305) main_task: Calling app_main()
I (305) uart: queue free spaces: 10
I (305) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (415) SIM_MODULE: Received data: AT
OK

I (415) SIM_MODULE: Got expected return value
I (415) SIM_MODULE: PCI module initialized
I (515) SIM_MODULE: Received data: AT+CGNSSPWR=1,1,1
OK

W (14515) SIM_MODULE: No response within timeout period.
W (28515) SIM_MODULE: No response within timeout period.
W (42515) SIM_MODULE: No response within timeout period.
```

Daniel-De-Dev, 2024-08-05:
Maybe the LEDs give a hint? I'm not sure as i cant find proper information about this either, Both red LEDs are solid, the blue one kind of oscillates in strength, not exactly on/off, just in strength.

lewisxhe, 2024-09-05:
Send `AT+CGNSSPWR=1` and wait for a while. Normally, it should return `+CGNSSPWR: READY!`  If it does not return, then you need to check whether the version you purchased has GPS.