Xinyuan-LilyGO/LilyGo-Camera-Series #46: Battery percentage and Deep sleep wakeup with battery

pol99b, 2024-08-30:
Dear fellows,

I have the T-SIMCAM OV2640 and the SIM7600G-H, which has the TP4056 battery charger, this charger only has a blue LED, I've seen other TP4056 that has a green and red a LED to know if the battery is charging or is fully charge. 
How can I know the charging voltage percentage or the remaining battery information? (I am using  18560 4400mAh 3.7V battery)

In addition, I am working with the deep sleep and it works perfect with the USB-C connected to my PC, waking up the module with a magnetic sensor. But when I plug a battery or a Table power supply when it enters the Deepsleep, the the battery supply shows current drop of the Deepsleep but then when I open the magnetic sensor it doesn't wake up. 
What it can be? How can I solve it?

lewisxhe, 2024-08-31:
https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series/blob/984166b0426e73df8551c0a9abc5e97c93a29649/examples/t_sim_cam_factory/t_sim_cam_factory.ino#L26
Is GPIO1 set high during the first step of setup?

pol99b, 2024-08-31:
> https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series/blob/984166b0426e73df8551c0a9abc5e97c93a29649/examples/t_sim_cam_factory/t_sim_cam_factory.ino#L26
> 
> 
> Is GPIO1 set high during the first step of setup?

Yes, this is the set up: 

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  pinMode(PWR_ON_PIN, OUTPUT);
  digitalWrite(PWR_ON_PIN, HIGH);

  Serial1.begin(115200, SERIAL_8N1, PCIE_RX_PIN, PCIE_TX_PIN);
  pinMode(PCIE_PWR_PIN, OUTPUT);
  digitalWrite(PCIE_PWR_PIN, 1);
  delay(500);
  digitalWrite(PCIE_PWR_PIN, 0);
  Serial.println("Wait...");
  delay(3000);

  pinMode(magneticSensorPin, INPUT_PULLUP);  // Configurar el pin del sensor magnético con resistencia interna pull-up

  Serial.println("Initializing modem...");
  delay(5000);  // Esperar a que el módem se estabilice

  if (!modem.init()) {
    Serial.println("Failed to initialize modem");
    return;
  }

  // Verificar que el módem esté registrado en la red
  if (modem.waitForNetwork()) {
    Serial.println("Network found");
  } else {
    Serial.println("Failed to find network");
    return;
  }

  String modemInfo = modem.getModemInfo();
  Serial.print("Modem Info: ");
  Serial.println(modemInfo);

  int res;
  do {
    res = modem.setNetworkMode(38);
    Serial.print("Modem Set LTE: ");
    Serial.println(res);
    delay(500);
  } while (res != 1);

  Serial1.println("AT+CNMP=38"); // Set the modem to LTE only mode
  Serial1.println("AT+CMNB=3");  // Set the modem to Cat-M and NB-IoT only
  delay(1000); // Wait for the command to take effect

  Serial1.println("AT+CPSI?"); // Get connection type and band
  delay(500);
  if (Serial1.available()) {
    String r = Serial1.readString();
    Serial.println(r);
  }

  // Configuración de wakeup
  esp_sleep_enable_ext0_wakeup(GPIO_NUM_21, 1);  // Configura el pin 21 para despertar el ESP32

  // Detecta la causa de despertar
  if (esp_sleep_get_wakeup_cause() == ESP_SLEEP_WAKEUP_EXT0) {
    delay(100);
    camera_init();
    takeShot();
  }

  Serial.println("Entrando en Deep-sleep...");
  delay(100);  // Asegura que todos los datos seriales se envíen antes de dormir
  esp_deep_sleep_start();
}

pol99b, 2024-09-05:
> Yes, this is the set up:
>
> void setup() { Serial.begin(115200); Serial.setDebugOutput(true); Serial.println();
>
> pinMode(PWR_ON_PIN, OUTPUT); digitalWrite(PWR_ON_PIN, HIGH);
>
> Serial1.begin(115200, [...]

Any solution? @lewisxhe

lewisxhe, 2024-09-05:
Are you using GPIO21 for wakeup? T-SIM GPIO21 doesn't seem to have an exposed pad. Are you sure you have connected it correctly?

pol99b, 2024-09-05:
> Are you using GPIO21 for wakeup? T-SIM GPIO21 doesn't seem to have an exposed pad. Are you sure you have connected it correctly?

Yes, I am using the GPIO21 for the wake up and it works perfectly with the USB-C, so it's connected correctly. The problem is when I connect the JST to a battery or to a adjustable power supply, the wake up does not work when the T-SIMCAM enters in deepsleep mode and it remains in deepsleep forever. 

What can I do?

[image]

lewisxhe, 2024-09-05:
GPIO21 is connected to the status indicator of the Modem. You can use the simplest sketch to read whether the reading changes after GPIO21 is connected to 3V3 or GND to determine whether it can be used.

pol99b, 2024-09-06:
> GPIO21 is connected to the status indicator of the Modem. You can use the simplest sketch to read whether the reading changes after GPIO21 is connected to 3V3 or GND to determine whether it can be used.

The reading changes after GPIO21 is connected to 3V3 or GND. But when it enters into deepsleep with a supply energy it does not wake up, with the USB-C connected to the PC it works perfectly but with the JST it does not wake up from the deepsleep. What it can be?

pol99b, 2024-09-06:
Is there any other RTC GPIO that is not connected to the modem that I can use in order to do the wake up from the deepsleep?

lewisxhe, 2024-09-09:
Have you tried unloading the modem first and then doing a separate deep sleep wake test?

pol99b, 2024-10-02:
[image]

What is this? I can't find what is it in the documentation?

Also Is there any possibility to connect a magnetic sensor in the T-SIMCAM?

lewisxhe, 2024-10-04:
For connecting cameras with IR filter function
[image]