Xinyuan-LilyGO/LilyGo-Camera-Series #47: PCie Sim dont wake up

JcortesSAI, 2024-10-01:
I've used the "t-sim-cam-factory" example from the repository, but my PCIe SIM7600 isn't waking up. I'm getting a strange character instead of 'OK'. I've tried the auto baud rate configuration, but it can't establish proper communication. Could my board be damaged? I'm using the T-SIMCAM 1.3.

i've made some modifications to see the answer from the PCIe 
Pinout
[image]
Method
[image]
Response 
[image]
"�"

lewisxhe, 2024-10-04:
Is the red light on the modem on?

JcortesSAI, 2024-10-04:
Nop, there is not

Enviado el: jueves, 3 de octubre de 2024 23:10
Para: Xinyuan-LilyGO/LilyGo-Camera-Series ***@***.***>
CC: JcortesSAI ***@***.***>; Author ***@***.***>

lewisxhe, 2024-10-05:
If the red light is not on, the modem is not powered on.

Did you add the power on pin in the setup initialization?
```
pinMode(PWR_ON_PIN, OUTPUT);
digitalWrite(PWR_ON_PIN, HIGH);
```