Xinyuan-LilyGO/LilyGo-Camera-Series #48: Serial monitor not responding

poison-cc, 2024-10-07:
I am using the OV5640, and I followed the instructions in the video https://www.youtube.com/watch?v=j9Z0KBuwEGI&t=1321s to upload the code. I waited for about an hour, but the serial monitor is not responding at all. 

I also tried uploading a simpler piece of code, but the serial monitor still doesn't display anything. 

I am not using the PCIE-SIM module, so I'm not sure if that has any effect.
[image]
[image]

lewisxhe, 2024-10-07:
Enable CDC
https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series/blob/984166b0426e73df8551c0a9abc5e97c93a29649/platformio.ini#L52

poison-cc, 2024-10-07:
I have enabled CDC, but the serial monitor still doesn't return anything
[image]

lewisxhe, 2024-10-07:
After starting up, wait a while and test again. As long as you can upload the sketch, it means there is no problem with the communication.
[image]