Xinyuan-LilyGO/LilyGo-Camera-Series #50: Remote streaming video

GiovanniDeRosa1, 2025-01-09:
Hi, i am using a T-SIMCAM with an a7670e module. I was able to stream video both from the ap and under the local network. Is it possible to stream video that can be accessed by users not under the same network ? 
Thanks for the support,
Giovanni

lewisxhe, 2025-01-10:
It can be accessed under the same network. You only need to initialize the specified STA mode. You can access it by entering the ESP IP address in the browser, but the video cannot be streamed under 4G.

lewisxhe, 2025-01-10:
The chip does not support video encoding. It can save pictures to SD. The essence of video is also each picture, but the frame rate is limited by the SD read and write speed and other speeds.

GiovanniDeRosa1, 2025-01-13:
Is there a lilygo solution that allows videos to be saved on an sd ?

lewisxhe, 2025-01-14:
We don't have one, but here is an example, you can refer to https://github.com/jameszah/ESP32-CAM-Video-Recorder

GiovanniDeRosa1, 2025-01-17:
the board has a microphone. is it possible to save audio files in mp3/wav to sdcard ? does it have a built-in codec?

lewisxhe, 2025-01-20:
Can save wav, esp official already has this example
https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP_I2S/examples/Record_to_WAV

GiovanniDeRosa1, 2025-01-21:
I ask for confirmation. Since the schematic does not show me psram, is missing ?

lewisxhe, 2025-01-22:
ESP32S3-R8 has built-in PSRAM. It exists inside the ESP32S3 chip.