Xinyuan-LilyGO/LilyGo-Camera-Series #51: Usage of esp-modem library with t-sim-cam

GuillaumeHupi, 2025-02-14:
Hello, 

We bought s3_cam_sim_v1.3 with pcie SOM7070G modem.

We try to use it to send big amount of data threw the modem in 1 message. But with AT commands we seem to be limited to 1024 caracter for MQTT and 4096 caracter for http body. 

So we tried to use the most recent esp-modem library with esp-idf with pppos library to try to send more data. With for example this example : 
https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/pppos_client

We modified the configuration to match the board. So we selected SIM7070 modem. Then changed TX and RX pin to match the one of our board. We also deactivated FLOW CONTROL since RST and DTR pins seems to be unavailable ont the pcie modem. Then added some lines to power up correctly the modem correctly on init. 

But we are facing errors ont init of the example. We encounter errors from the first communications with the modem.

Even the esp_modem_get_signal_quality() is not able to work : 

I (360) pppos_example: Power on the modem
E (4360) pppos_example: esp_modem_get_signal_quality failed with 263 ESP_ERR_TIMEOUT
I (4360) main_task: Returned from app_main()

I specify that we managed to make the modem work through AT commands and to send data in MQTT via these commands. So the modem works.
So I don't know if we are configuring the modem wrong for the esp-modem library or is we just can't use your hardware to work with this library. Your help on this topic will be appreciated. Maybe you have examples of using this modem with the esp-modem library, or with the pppos commands? Or on the contrary that you have already tried and you know that it cannot work for hardware reasons.

Thanks
Guillaume

barbiani, 2025-04-01:
I did use the arduino version and it worked. Did you check the modem tx pin for the returned data?

Also, did you write HIGH to gpio 1 to power up the pcie slot?

GuillaumeHupi, 2025-04-22:
Hi @barbiani, 

Yes I set the GPIO 1 to high on boot. 
Which library did you use ? I tried with this one : 

https://github.com/espressif/arduino-esp32/blob/master/libraries/PPP/examples/PPP_Basic/PPP_Basic.ino

I modified it a little bit the code to work with my sim card and the wiring of the board. 

Here is the sketch i used : 

``` C
#include <PPP.h>

#define PPP_MODEM_APN "iot.1nce.net"
#define PPP_MODEM_PIN NULL  // or NULL

// WaveShare SIM7600 HW Flow Control
// #define PPP_MODEM_RST       25
// #define PPP_MODEM_RST_LOW   false  //active HIGH
// #define PPP_MODEM_RST_DELAY 200
// #define PPP_MODEM_TX        21
// #define PPP_MODEM_RX        22
// #define PPP_MODEM_RTS       26
// #define PPP_MODEM_CTS       27
// #define PPP_MODEM_FC        ESP_MODEM_FLOW_CONTROL_HW
// #define PPP_MODEM_MODEL     PPP_MODEM_SIM7600

// SIM800 basic module with just TX,RX and RST
#define PPP_MODEM_RST     48
#define PPP_MODEM_RST_LOW false //active LOW
#define PPP_MODEM_TX      45
#define PPP_MODEM_RX      46
#define PPP_MODEM_RTS     -1
#define PPP_MODEM_CTS     -1
#define PPP_MODEM_FC      ESP_MODEM_FLOW_CONTROL_NONE
#define PPP_MODEM_MODEL   PPP_MODEM_SIM7070
#define PPP_MODEM_RST_DELAY 2000

#define PWR_ON_PIN 1

void onEvent(arduino_event_id_t event, arduino_event_info_t info) {
  switch (event) {
    case ARDUINO_EVENT_PPP_START:        Serial.println("PPP Started"); break;
    case ARDUINO_EVENT_PPP_CONNECTED:    Serial.println("PPP Connected"); break;
    case ARDUINO_EVENT_PPP_GOT_IP:       Serial.println("PPP Got IP"); break;
    case ARDUINO_EVENT_PPP_LOST_IP:      Serial.println("PPP Lost IP"); break;
    case ARDUINO_EVENT_PPP_DISCONNECTED: Serial.println("PPP Disconnected"); break;
    case ARDUINO_EVENT_PPP_STOP:         Serial.println("PPP Stopped"); break;
    default:                             break;
  }
}

void testClient(const char *host, uint16_t port) {
  NetworkClient client;
  if (!client.connect(host, port)) {
    Serial.println("Connection Failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    client.read();  //Serial.write(client.read());
  }

  Serial.println("Connection Success");
  client.stop();
}

void setup() {
  Serial.begin(115200);
  pinMode(PWR_ON_PIN, OUTPUT);
  digitalWrite(PWR_ON_PIN, HIGH);
  delay(5000);
  Serial.println("Starting the example");
  // Listen for modem events
  Network.onEvent(onEvent);

  // Configure the modem
  PPP.setApn(PPP_MODEM_APN);
  PPP.setPin(PPP_MODEM_PIN);
  PPP.setResetPin(PPP_MODEM_RST, PPP_MODEM_RST_LOW, PPP_MODEM_RST_DELAY);
  PPP.setPins(PPP_MODEM_TX, PPP_MODEM_RX, PPP_MODEM_RTS, PPP_MODEM_CTS, PPP_MODEM_FC);

  Serial.println("Starting the modem. It might take a while!");
  PPP.begin(PPP_MODEM_MODEL);

  Serial.print("Manufacturer: ");
  Serial.println(PPP.cmd("AT+CGMI", 10000));
  Serial.print("Model: ");
  Serial.println(PPP.moduleName());
  Serial.print("IMEI: ");
  Serial.println(PPP.IMEI());

  bool attached = PPP.attached();
  if (!attached) {
    int i = 0;
    unsigned int s = millis();
    Serial.print("Waiting to connect to network");
    while (!attached && ((++i) < 600)) {
      Serial.print(".");
      delay(100);
      attached = PPP.attached();
    }
    Serial.print((millis() - s) / 1000.0, 1);
    Serial.println("s");
    attached = PPP.attached();
  }

  Serial.print("Attached: ");
  Serial.println(attached);
  Serial.print("State: ");
  Serial.println(PPP.radioState());
  if (attached) {
    Serial.print("Operator: ");
    Serial.println(PPP.operatorName());
    Serial.print("IMSI: ");
    Serial.println(PPP.IMSI());
    Serial.print("RSSI: ");
    Serial.println(PPP.RSSI());
    int ber = PPP.BER();
    if (ber > 0) {
      Serial.print("BER: ");
      Serial.println(ber);
      Serial.print("NetMode: ");
      Serial.println(PPP.networkMode());
    }

    Serial.println("Switching to data mode...");
    PPP.mode(ESP_MODEM_MODE_CMUX);  // Data and Command mixed mode
    if (!PPP.waitStatusBits(ESP_NETIF_CONNECTED_BIT, 1000)) {
      Serial.println("Failed to connect to internet!");
    } else {
      Serial.println("Connected to internet!");
    }
  } else {
    Serial.println("Failed to connect to network!");
  }
}

void loop() {
  if (PPP.connected()) {
    testClient("google.com", 80);
  }
  delay(20000);
}
```

Like this, it seems to work ! Thanks ! 

So now that I know ppos is working with arduino, I will try to understand why I wasn't able to make it work with esp-idf.